rzeldent/platformio-espressif32-sunton #24: Pin mapping for CFCard I/F

vedecoid, 2024-01-03:
I was going through the pin mappings for the SDcard I/F on the  7" and 4" 800x480 displays (ESP32-8048S043-C and ESP32-8048S070-C) and noticed that they do not comply with the schematic for these boards. In the config files, these mappings are used
      "-DTF_CS=5",
      "-DTF_SPI_MOSI=23",
      "-DTF_SPI_SCLK=18",
      "-DTF_SPI_MISO=19"
while the schematic shows:
CS : IO10
MOSI : IO11
CLK : IO12
MISO : IO13

rzeldent, 2024-01-03:
Thanks! Has been updated!

closed 2024-01-03 by rzeldent

vedecoid, 2024-01-04:
:-)
I only checked the 2 boards I own though...

rzeldent, 2024-01-04:
The schematics are in the repo and checked; all these never boards have this IO assignment!
Thanks again for mentioning.