rzeldent/platformio-espressif32-sunton #37: not work installation board

herry93, 2024-02-18:
PIO Core Call Error: "The following files/directories have been created in D:\project\esp32-s3 tft\master\ver 2 LVGA\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving esp32-4827S043C dependencies...\r\n\n\nUnknownBoard: Unkn
[image]
own board ID 'esp32-4827S043C'"

rzeldent, 2024-02-18:
You need to checkout with submodules. The boards directory will then contain the boards

rzeldent, 2024-02-24:
Assume this has been resolved...

closed 2024-02-24 by rzeldent