rzeldent/platformio-espressif32-sunton #42: esp32-8048S050C `R` and `B` channels swaped

h3lgi, 2024-02-24:
The board definition corresponds to the schematic and I also double-checked by tracing the PCB traces, but it looks like on the actual display they are swapped or gpio labels on the schematic by sunton is just wrong. 

Display info:
board ID: 8048S050C from the URL given on the board definition JSON file
board version: V1.2 

[image]

This is the experimental UI in SquareLine Studio:

[image]

This is the result with the current config.
[image]

This is the result if I swapped `R` and `B` channels.

[image]

Here is the definition of the swapped channels that showing the correct colors.

```json
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R0=8'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R1=3'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R2=46'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R3=9'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R4=1'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G0=5'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G1=6'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G2=7'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G3=15'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G4=16'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G5=4'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B0=45'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B1=48'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B2=47'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B3=12'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B4=14'",
```

P.S. I saw that the board definition is updated to correspond to the schematic but yah looks like magic happens here :D

rzeldent, 2024-02-24:
Hi Oleg,

I've just updated in the develop!

h3lgi, 2024-02-25:
Hi 
Since this is specifically on my two versions of the display I am not sure if this will be valid for all of them, so I just shared my observation. 

In any case, I think it will be ok to close this issue so if anybody experiences any issues with colors they can take a look at this issue or to this one: #35 

@rzeldent thank you for such a quick reaction and amazing job

closed 2024-02-25 by h3lgi