rzeldent/platformio-espressif32-sunton #43: Touch Error

gathualex, 2024-02-28:
Building from develop branch for ESP32-8048S050C

`.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/lvgl_gt911_i2c.c:72:30: error: 'LCD_SWAP_XY' undeclared (first use in this function); did you mean 'TOUCH_SWAP_XY'?

         .flags = {.swap_xy = LCD_SWAP_XY, .mirror_x = LCD_MIRROR_X, .mirror_y = LCD_MIRROR_Y},
                              ^~~~~~~~~~~
                              TOUCH_SWAP_XY
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/lvgl_gt911_i2c.c:72:30: note: each undeclared identifier is reported only once for each function it appears in
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/lvgl_gt911_i2c.c:72:55: error: 'LCD_MIRROR_X' undeclared (first use in this function)
         .flags = {.swap_xy = LCD_SWAP_XY, .mirror_x = LCD_MIRROR_X, .mirror_y = LCD_MIRROR_Y},
                                                       ^~~~~~~~~~~~
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/lvgl_gt911_i2c.c:72:81: error: 'LCD_MIRROR_Y' undeclared (first use in this function)
         .flags = {.swap_xy = LCD_SWAP_XY, .mirror_x = LCD_MIRROR_X, .mirror_y = LCD_MIRROR_Y},
                                                                                 ^~~~~~~~~~~~
*** [.pio\build\esp32-8048S050C\libe09\esp32_smartdisplay\lvgl_gt911_i2c.c.o] Error 1
================================================================================== [FAILED] Took 212.73 seconds ==================================================================================
`

rzeldent, 2024-02-28:
Hi Alex,

This is currently refactored. Can you do an update or switch to main?

rzeldent, 2024-03-17:
Assume has been fixed...

closed 2024-03-17 by rzeldent