rzeldent/platformio-espressif32-sunton #51: [ESP32-2432S024C] uses CST820 instead of CST816 and the INT GPIO is different

regystro, 2024-04-11:
Hi. I recently bought an ESP32-2432S024C, and can confirm that there are some changes regarding board definition: https://github.com/rzeldent/platformio-espressif32-sunton/blob/main/esp32-2432S024C.json

1. The touch sensor is a CST820 instead of `"'-D TOUCH_CST816S_I2C'"`
2. The Interrupt GPIO is 22 instead of `"'-D CST816S_TOUCH_CONFIG_INT_GPIO_NUM=21'"`

I tested it with https://esphome.io/components/touchscreen/cst816.html and works as expected.

rzeldent, 2024-04-11:
Hi Regystro,

Thanks for going into the depths of the smartdisplay library!

It is very well possible the sensor is a CST820. The library CST816S is upwards compatible so supports this touch screen chip. Making a new library for the same functionality is not worth the effort so will stick with this module.

The INT GPIO is not used; it is polled by LVGL. So this is not much of an issue but the definitions should be correct!
I checked the schematic:
[image]
And it seems to be 21. Where did you find this information or did you measure/followed the tracks it on your board?

Kind regards,

Rene

regystro, 2024-04-11:
Hi Rene.

Regarding the INT GPIO, I have the same schematics and even the sample code uses GPIO21. Anyway, trying to get it to work with ESPHome -which can use interrupts and also polling- I found that disabling polling, only GPIO22 detects interrupts. It was a matter of trial an error. GPIO22 shows touch in the logs, while GPIO21 doesn't.

It could be a mismatch in ESPHome's GPIO assignation though :)

Regards.

rzeldent, 2024-04-16:
It is very well possible this this the case and they a really good find...

How can I reproduce this? Is there a little demo app to check for the INT

regystro, 2024-04-17:
I digged deeper and found that even setting an interrupt pin in ESPHome, which as per the documentation should stop using the polling method (= `update_interval: never`), with CST820 it's not working.

Long story short: I thought I was using interrupts while the code was using polling. I filled an issue to see if ESPHome developers can fix it: https://github.com/esphome/issues/issues/5708

rzeldent, 2024-04-17:
Okay, thanks for looking into this issue...
So might be correct after all...

Arunce159, 2024-11-20:
Hello, i am working on the same display, only programming on arduino and i cant get that touch to work, could you help me?

rtek1000, 2025-06-24:
Hello, the driver for CST820 is very simple, unfortunately I haven't found a complete datasheet to know how to activate the Interrupt pin (which is an output and must be read from a microcontroller input).

I found the driver for a 2.4 inch display, and it is available in this repository:

https://github.com/rtek1000/2.4inch_ESP32-2432S024

[CST820_driver_cpp.zip](https://github.com/user-attachments/files/20889802/CST820_driver_cpp.zip)