rzeldent/platformio-espressif32-sunton #61: Blank screen on ESP32-3248S035 capacitive

dcknuth, 2024-08-29:
Got everything to compile and upload, but just a blank screen. Tried changing:
#define LV_USE_ST7796		1
in lv_config.h, but seems ineffective. I set the default board with:
default_envs = esp32-3248S035C
and that part seemed fine.

The preloaded demo worked when powering on the board, so I assume the board itself is OK.

rzeldent, 2024-09-01:
Which project are you using? If it is the [l](https://github.com/rzeldent/esp32-smartdisplay) project, it is just a library and has no output. If so, try the https://github.com/rzeldent/esp32-smartdisplay-demo project!

dcknuth, 2024-09-01:
It's the *-demo one. I was able to get some of the provided demos from Sunton to work after realizing that one has to set all #defines as command line switches. I noticed in a note on the CYD support pages you may have hacked your copy of this particular board for a different refresh rate. I also may have to learn how to get a crash dump or some output on my run of your demo. Looks like there might be some info there.

dcknuth, 2024-09-01:
Just got a esp32-8048S043C and your demo worked out-of-the-box on that one. Very nice